ubc-aamodt-group/vulkan-sim #2: Error when run RTV6

liqingy5, 2022-10-04:
Hello,
I was following the instructions for building the vulkan-sim and mesa, everything is looking good but when I tried to run the RTV6 to dump the trace, I ran  ./RayTracer and got this X error of failed request.
[image]

I was using Ubuntu 20.04 with CUDA 11.7 ,gcc /g++ 9.4.0 and GPU is GTX 1080, I can run the RTV6 without involving vulkan-sim, any idea what might cause this error?

tommychouyc, 2022-10-04:
Does the computer you're running off have Intel integrated graphics? We currently don't support generating traces with Nvidia GPUs.
Also if you could provide a screenshot of the gdb backtrace, that will be very helpful. Thanks!

liqingy5, 2022-10-06:
Sorry for the late reply, it's probably the issue of Xorg on my local machine, I will look into that first when I had the chance (quite busy recently). By the way, can Vulkan-sim be installed on other Linux OS like centOS?

tommychouyc, 2022-10-06:
We haven't tried it out before, but it's probably compatible. Please let us know how it goes!

closed 2022-10-24 by tommychouyc