ubc-aamodt-group/vulkan-sim #4: Does vulkan-sim support run application directly ?

xiexinghua, 2022-10-12:


tommychouyc, 2022-10-12:
Yes it does. Currently we support running a Vulkan ray tracing application on systems with an Intel CPU with integrated graphics as stated in the README. Vulkan-Sim simulates the ray tracing part of the application directly. It does not run off the integrated graphics, but it's just a requirement for mesa to work.

For more instructions on how to do so, please refer to [General Instructions For Trace Dumping](https://github.com/ubc-aamodt-group/vulkan-sim#general-instructions-for-trace-dumping) in the README. Also rememeber to set $VK_ICD_FILENAMES to the location of your intel_icd.x86_64.json in mesa so the Vulkan application uses the Vulkan-Sim's driver.

xiexinghua, 2022-10-12:
I have a question， if the vulkan-sim can run application directly，why we need integrated graphics.  As we know, the real software stack for run app  A ： software rendering    app + mesa + cpu；  B ：GPU rendering  app + mesa + libdrm + DRM driver + gpu.  what  kind   of the vulkan-sim?  so I don't know why the vulkan-sim need dump the trace for replay.

tommychouyc, 2022-10-12:
We only simulate the ray tracing part of the Vulkan application by intercepting certain API calls such as vkCmdTraceRaysKHR. The other API calls are still handled by the mesa driver such as windowing, creating the vkInstance, and handling API calls that query the vkPhysicalDevice, etc.

The Intel-Mesa driver that we built off only had ray tracing shader compiler support but did not have functionality to execute ray tracing shaders. We extended this by adding ray tracing functionality inside Vulkan-Sim's functional simulation. As a result, the Intel-mesa driver requires the system to have Intel integrated graphics. We hope to resolve this limitation in the future and welcome contributions.

The reason we have traces is so the user can dump traces from a supported machine (e.g. an Intel laptop), then perform experiments on a server/cluster that is much more powerful but doesn't have integrated graphics.

closed 2022-10-24 by tommychouyc