ubc-aamodt-group/vulkan-sim #5: Error while running ./RayTracer --scene 6 --height 320 --width 448

pratyush48, 2023-02-10:
The build of mesa and vulkan-sim is successful. While running the Raytracer. I am getting AttributeError: 'NoneType' object has no attribute 'isLoadConst'. I have attached the error screenshot after printing output of line.fullLine. Kindly have a look at it. 

[image]

mohammadrezasaed, 2023-02-11:
Is this RayTracingInVulkan workload? If you are using the original version please try [this ](https://github.com/ubc-aamodt-group/RayTracingInVulkan)one.

Can you send us the contents of the "mesagpgpusimShaders" folder? It holds all the shader codes.
Do you know which shader is causing the error?

pratyush48, 2023-02-11:
Hi,
I am using the same workload mentioned in the build and I think the mail
also refers to the same one. I have attached the screenshot of the contents
available in the "mesagpgpusimShaders" for your reference. I am also
attaching a screenshot of the output which mentions it is the "RAYGEN"
shader causing the error.
[image: image.png]
[image: image.png]

pratyush48, 2023-02-11:
Hi,
I am using the same workload mentioned in the build and I think the mail
also refers to the same one. I have attached the screenshot of the contents
available in the "mesagpgpusimShaders" for your reference. I am also
attaching a screenshot of the output which mentions it is the "RAYGEN"
shader causing the error.

mohammadrezasaed, 2023-02-11:
I can't see the images you've attached.
Can you attach the RAYGEN shader file from the "mesagpgpusimShaders"?

pratyush48, 2023-02-12:
It's probably because I sent it via mail. I will attach the images here again. The RAYGEN shader had a ptx format which is not supported by github so I have changed it to .txt to attach it. Kindly have a look at it.

[MESA_SHADER_RAYGEN_0.txt](https://github.com/ubc-aamodt-group/vulkan-sim/files/10717270/MESA_SHADER_RAYGEN_0.txt)
[image]
[image]

mohammadrezasaed, 2023-02-15:
There seem to be some NIR instructions generated by Mesa that we don't currently support in Vulkan-Sim. I'm not sure why the generated NIR in your system is different from ours. Might be because you have a different intel GPU.

These instructions are refered to as "Untranslated NIR instruction" in the shader file. All the untranslated ones seem to be ALU instructions. You can use the NIR ALU [documentation](https://docs.mesa3d.org/nir/alu.html) and add NIR to PTX translation for these instructions [here](https://github.com/ubc-aamodt-group/mesa-vulkan-sim/blob/5bbedf981268bf80cdee065372d89a1912759df1/src/compiler/nir/nir_print.c#L2494).

pratyush48, 2023-02-16:
I have an intel integrated graphics card [Mesa Intel® UHD Graphics (ICL GT1)]. Can the reason be due to the old intel mesa drivers? I can try to download the latest open-source mesa drivers.

mohammadrezasaed, 2023-02-17:
I'm not sure if upgrading mesa would help. You also have to integrate the changes we made to mesa.
We generated the NIR with "Intel(R) HD Graphics 630 (KBL GT2)". You can find how mesa is differentiating between the two GPUs and force it to use the same compilation path as the 630 GPU or implement the new instructions.

closed 2023-03-10 by mohammadrezasaed