ubc-aamodt-group/vulkan-sim #7: Error while running ./RayTracer --scene 6 --height 320 --width 448

Cuicunhao, 2023-04-04:
The build of mesa and vulkan-sim is successful. While running the Raytracer, I am getting symbol lookup error: '~/vulkan-sim-root/mesa-vulkan-sim/lib/lib/x86_64-linux-gnu/libvulkan_intel.so: undefined symbol: gpgpusim_setGeometries'. Please have a look at it.
[image]

tommychouyc, 2023-04-04:
Hi,

Could you modify a few lines in this file? https://github.com/ubc-aamodt-group/mesa-vulkan-sim/blob/main/src/intel/vulkan/meson.build

1. Comment out lines 30 and 31.
2. Uncomment lines 29, 32, and 33, then modify the path in line 29 to be the libcudart.so of the vulkan-sim or gpgpusim library path.
3. You probably should also modify the embree_lib_dir and embree_header_dir paths in lines 22 and 23 of the file as well.
4. Recompile both mesa and vulkan-sim.
5. Try running the workload again.

Please let me know if the fix works!

Cuicunhao, 2023-04-05:
> Hi,
>
> Could you modify a few lines in this file? https://github.com/ubc-aamodt-group/mesa-vulkan-sim/blob/main/src/intel/vulkan/meson.build
>
>     1. Comment out lines 30 and 31.
>
>     2. Uncomment [...]

Thank you very much! The fix works! I noticed that my mistake was to change the path on line 29 to libcudart.so in cuda11.1........

closed 2023-04-05 by Cuicunhao