ubc-aamodt-group/vulkan-sim #8: A segmentation fault happens when running the "./vulkan_rt_runner RTV6-trace/ RTV6"

Cuicunhao, 2023-04-06:
Hello,
when I tried to run the "./vulkan_rt_runner RTV6-trace/ RTV6, I got a segmentation fault. When I followed the tutorial instructions and rerun using gdb, I found that it crashes at `VulkanRayTracing::traceRay`. 
[image]
I'm not sure if this issue is due to not waiting for 'Trace dumped' to appear in the terminal when running RTV6. Because I did not find 'Trace dumped' in the terminal, but the terminal displayed the interface shown in the following figure, I felt that the program had already run. Could you please help me solve the segmentation fault? Thanks a lot.
[image]

mohammadrezasaed, 2023-04-25:
It seems the BVH tree is invalid and you get an error when traceRay function tries to read it.
Did you change the `use_external_launcher` to `false` when dumping the trace?
Can you check the files in your dumped trace? The files should look similar to [this](https://github.com/ubc-aamodt-group/trace-runner-vulkan-sim/tree/main/RayTracingInVulkan-Scene6-448x320-Baseline).

MascaChapas27, 2023-06-30:
The same thing is happening to me. I changed `use_external_launcher` to `false` and commented out the `continue;`s, and then compiled using `make -j`, and after executing `./RayTracer` the files in my dumped trace look like this:

[image]

Also, when executing `./RayTracer`, the screen is completely black when the Ray Tracing option is enabled, but the scene can be seen when the option is disabled.

Cuicunhao, 2023-09-08:
Thanks a lot, I'm sorry for the delayed response. I will attempt to use Vulkan-Sim 2.0.0 and close this issue.

closed 2023-09-08 by Cuicunhao