ubc-aamodt-group/vulkan-sim #9: Error while running other traces

pratyush48, 2023-04-12:
1. **While running other traces:**  

- https://github.com/GPSnoopy/RayTracingInVulkan

[image]

- https://github.com/KhronosGroup/Vulkan-Samples

[image]

I am getting these errors. Kindly let me know how I can solve these errors.

2. **After dumping the traces**(https://github.com/ubc-aamodt-group/RayTracingInVulkan) takes a very long time to run and at the end gives a segmentation fault stating the root node is unable to access an address. Is there any way I can run the traces in batch mode or limited to an area so that the time to run traces decrease significantly?

pratyush48, 2023-04-26:
The traces you provided work. I changed the resolution after dumping and the time it took to render was significantly less. Can you suggest to me which font I need to download in order to run Vulkansamples?
For RayTracingVulkan when I removed the --fullscreen argument it still gave the same error, is there anything else I am missing?

mohammadrezasaed, 2023-04-26:
In the image you provided I can see there's an error with not finding a font in the assets/fonts folder.
Also note that you should execute Vulkan-Samples from the parent `Vulkan-Samples` folder and not the folder that contains the binary file.
`./build/linux/app/bin/Debug/x86_64/vulkan_samples sample raytracing_extended`

For RayTracingInVulkan I can see an error with not finding the compute queue. Are you using our version of [RayTracingInVulkan](https://github.com/ubc-aamodt-group/RayTracingInVulkan)? We removed the compute queue from the original version since it wasn't being used.

closed 2023-05-30 by mohammadrezasaed