ubc-aamodt-group/vulkan-sim #11: RTV6 cannot find a suitable device?

yavuz650, 2023-06-19:
I installed vulkan-sim in a docker container along with dependencies. When I run the ray tracing app, the integrated Intel GPU is picked as the physical device, however, it lacks the "VK_KHR_ray_tracing_pipeline" extension, so the app throws an error (main.cpp:243). 

Integrated Intel GPUs do not have hardware acceleration for ray tracing, so I'm not sure why the app is looking for the extension. 

Any hints?

I'm running vulkan-sim in a Ubuntu 20.04 container with Cuda 11.7 with an Intel 12700H CPU.

JRPan, 2023-07-09:
I'm having a similar issue with a 13th gen Intel CPU. I think the Mesa driver needs to be updated to support the latest Intel GPUs. Can someone confirm that? And is there any plan for that? 

Thanks!

mohammadrezasaed, 2023-07-18:
The version of Mesa in Vulkan-Sim doesn't support 13th gen Intel CPU. We will release a version of Vulkan-Sim that doesn't need intel integrated GPUs.

tommychouyc, 2023-08-30:
We've just released Vulkan-Sim 2.0.0 and the corresponding mesa repo which removes the Intel driver dependency and can now be run directly on any CPU.

closed 2024-05-09 by tommychouyc