ubc-aamodt-group/vulkan-sim #15: Segmentation fault while running vulkan-sim 2.0.0

GETINSIDE3, 2023-10-10:
Hello,
when I try to run Lumibench on vulkan-sim 2.0.0, the program exits with a segmentation.
I've tried to run it on various settings (e.g., different servers, packages, scenes and resolutions), but similar errors consistently occurred.
The tested environments are as follows.
CPU: EPYC 7551, Xeon E5-2680v4, Core i7-13700K
OS: Ubuntu 20.04
CUDA: 11.7
Embree: v3.13.4, v3.12.2
Vulkan SDK: 1.3.216, 1.3.261

Here's the backtraces that I obtained from gdb.
I guess it might be something related to intersim2, but I couldn't really figure out the source of the error.
I would greatly appreciate it if you could kindly look at it.

[image]

quanzw99, 2023-10-10:
I had the same situation.

I don't have much experience using gdb or other tools to find more error information. But I followed most of the steps. However, I still received a segmentation fault result.

Additionally, I would like to confirm if there have been any changes to the configuration files for `embree_lib_dir` and `gpgpusim_lib_dir`. Is it the right configuration time after generating libcudart.so in /vulkan-sim/lib/?

[image]

lucylufei, 2023-10-10:
Hi @GETINSIDE3, could you please attach the `gpgpusim.config` file you used and the full terminal output?

@quanzw99, there should be no change for the configuration file between Vulkan-Sim 1.0.0 and 2.0.0. I believe your configurations are correct for `embree_lib_dir` and `gpgpusim_lib_dir` based on your attached output. 
Please try running with gdb to see if your segmentation fault matches the other as well. You can do this by adding `gdb --args` before the application command. For example: `gdb --args ./RayTracer --scene 1 --width 32 --height 32 --samples 1`. When the segmentation fault occurs, type `bt` to get the backtrace information. Thanks!

quanzw99, 2023-10-10:
@lucylufei  Thanks for your prompt reply.

I followed your instructions and ran the program with GDB. Here's a screenshot of the results.

[image]

anaitha2, 2023-10-10:
Hi, I keep encountering the same issue for all scenes as well.

[image]

From what I debugged the issue seems to be in executing the push in file vulkan-sim-root/vulkan-sim/src/cuda-sim/vulkan_ray_tracing.cc

(line 1598) ctx->the_gpgpusim->g_stream_manager->push(op);

lucylufei, 2023-10-10:
Hi everyone, thank you so much for providing details on this segmentation fault. 

Please pull the latest commit and try the updated version of the `gpgpusim.config` file that reverts the interconnect settings. To be more specific, pull this vulkan-sim repo, then redo step 3 of "Running RayTracingInVulkan (LumiBench) with Vulkan-Sim" from the README. This _should_ prevent the segmentation fault and we will investigate further regarding the problematic configurations. 

Thanks!

quanzw99, 2023-10-10:
@lucylufei After following these steps, the segmentation fault indeed disappeared. However, it seems to be stuck at the `saving image SCENE to file` stage. When I tried to open the PPM file using the PPM viewer, it says that `Not enough pixel data`. Is this the expected result for us at this point?

Thank you for your time and assistance!

[image]

quanzw99, 2023-10-12:
> @lucylufei Thanks for the prompt feedback! I have followed the instructions and the simulator seems to be working just fine so far.
> 
> @quanzw99 Mine was stuck for quite a long time in the same stage, when I ran ./RayTracer --scene 18 --width 128 --height 128. But I found that I omitted --samples option. When I ran the same benchmark with ./RayTracer --scene 18 --width 128 --height 128 --samples 1, it worked just fine and took about 15mins of time. (Simulation for the same scene with samples=2 took 26mins for my case). I guess it might have been the simulation itself taking too long since it seems like the default value of the samples parameter is set to 8 unless specified.

@GETINSIDE3 Thank you for your reply! After waiting for a while, the complete ppm file was indeed generated. I used the `./RayTracer --scene 18 --width 128 --height 128 --samples 2` to generate the following image (opened with a ppm viewer). Is this the expected output?

[image]

closed 2023-10-13 by GETINSIDE3