ubc-aamodt-group/vulkan-sim #16: Wrong Descriptor Set index

JRPan, 2023-10-10:
When running a large project with more than one descriptor set is used, the index is always 0 in the PTX code. So my textures are just wrong. Cannot reference the correct descriptor set because of this.
The associated code is below:
https://github.com/ubc-aamodt-group/mesa-vulkan-sim/blob/b8f3e3980e87736fea176da6ebc81f82c6ea80e9/src/compiler/nir/nir_print.c#L5121-L5122

```
fprintf(fp, "decl_var %s, %d, %d, %d, %d, %u, %u;\t", var->name, size, glsl_get_vector_elements(var->type), 
                  glsl_get_base_type(var->type), var->data.mode, var->data.driver_location, var->data.binding);
```

The descriptor set index is `var->data.driver_location`. However, at least for me, `var->data.descriptor_set` has the correct descriptor set index and should be used.
Is this a bug? Why is `driver_location` used? I know there are other variable types. Do other variables use driver_location instead of `var->data.descriptor_set`? Is it safe to change `var->data.driver_location` to `var->data.descriptor_set` directly?

Thanks,
JR

JRPan, 2023-10-10:
FYI
I extended Vulkan-sim a lot and added tons of features. Maybe this is something that you did not support. I just want to know if this `var->data.driver_location` is used anywhere else (Other than as a descriptor set index).

lucylufei, 2023-10-12:
We have not tried any large projects with more than one descriptor set. I believe you are correct to change `var->data.driver_location` to `var->data.descriptor_set` if you find that `var->data.descriptor_set` holds the descriptor set ID value. 

The `var->data.driver_location` was likely used following the code originally in nir_print.c from Mesa. However, for Vulkan-Sim, the generated file is immediately post-processed by ptx_lower_instructions.py, where it appears that the `var->data.driver_location` is only being used as the set ID. 

Thank you for extending Vulkan-Sim! Please create a pull-request if there is any work you would like to share :)

JRPan, 2023-10-12:
Sounds good. Thank you very much!

closed 2023-10-12 by JRPan