ubc-aamodt-group/vulkan-sim #17: Scene 12 (PartyTug) generates an empty image

angelms2003, 2023-11-15:
Hello, I have been trying the new vulkan-sim version with Lumibench and everything works perfectly, except for scene 12. After executing Ray Tracer with scene 12, default shader type, a size of 200x200 and 5 samples per pixel, the generated image is the following:
[image]

I tried generating the same image using the shadow shader with 5 samples per pixel and 5 shadow rays cast per ray, with the following result:
[image]

Also, for scene 13 (FOX) the head model was split in many small files, so I had to merge them. I thought the problem would be the same in this case, but it doesn't seem like that.

Also, when I opened the models for this scene in Blender, they had a weird position under the floor. Maybe this is causing the image to be empty? For example, this is _Grill_001.obj:

[image]

lucylufei, 2023-11-15:
Hello, I am not able to reproduce this issue on my end and this is the output for 200x200 5spp scene 12 that I see. 
[image]

Could you please try to run this scene using a physical GPU to confirm that the object files are not corrupted somehow? Also, please try to re-clone these repositories to make sure all the files match. We have a Docker container available on [Zenodo](https://zenodo.org/records/8267898) that you can try as well. If none of these options work, please upload the Vulkan-Sim console output.

As a side note, you can change the `-gpgpu_ptx_sim_mode` flag to `1` in the gpgpusim.config configuration file to run the functional simulation only, which is useful for debugging because it is much faster and still generates the ppm image. 

Thanks!

angelms2003, 2023-11-29:
Hello again. Unfortunately, I don't have a GPU that supports ray tracing in order to try the scene in real hardware. I tried using the docker container on a server with two Intel(R) Xeon(R) Gold 5218 CPU @ 2.30GHz processors and Ubuntu 20.04.5 LTS, but the generated image was the same. The output was the following:

[PARTY-server.log](https://github.com/ubc-aamodt-group/vulkan-sim/files/13503673/PARTY-server.log)

Also, I tried running it on my laptop, which has one Intel(R) Core(TM) i7-8565U CPU @ 1.80GHz processor and Ubuntu 22.04.3 LTS, and the image was generated correctly. The output was the following:

[PARTY-laptop.log](https://github.com/ubc-aamodt-group/vulkan-sim/files/13503704/PARTY-laptop.log)

Thank you for your help!

lucylufei, 2023-12-01:
That's very odd. From your PARTY-server.log, I see some differences with Mesa and translating the shaders to PTX. Could you please share your .ptx files? They should be generated in `/home/vulkansim/mesa/gpgpusimShaders` in the Docker container. 

Also, your PARTY-laptop.log doesn't seem to have the actual output. Could you please double-check and re-upload? 

As another debugging test, could you run the full performance simulation (with `-gpgpu_ptx_sim_mode ` set to `0`) on the server and on your laptop and compare the `gpu_sim_cycle` value? You can run it with a smaller resolution to make it faster, as long as it's the same resolution on both the laptop and server. This will help us determine whether something is wrong with the shader code. 

Thanks!

angelms2003, 2023-12-02:
Sorry! It seems like the wrong file was sent. The actual output is the following:

[PARTY-laptop.log](https://github.com/ubc-aamodt-group/vulkan-sim/files/13537034/PARTY-laptop.log)

The .ptx files generated by the server are the following:

[ptx_files.zip](https://github.com/ubc-aamodt-group/vulkan-sim/files/13537060/ptx_files.zip)

Also, when executing `./RayTracer --scene 12 --height 100 --width 100 --samples 3` with `-gpgpu_ptx_sim_mode` set to `0`, the results are the following:

[PARTY-laptop-100x100.log](https://github.com/ubc-aamodt-group/vulkan-sim/files/13537042/PARTY-laptop-100x100.log)
[PARTY-server-100x100.log](https://github.com/ubc-aamodt-group/vulkan-sim/files/13537048/PARTY-server-100x100.log)