ubc-aamodt-group/vulkan-sim #20: Error while running Lumibench

anaitha2, 2024-01-30:
Hi,

I've freshly setup Vulkansim and Lumibench. While running Lumibench I'm encountering the following error:

./RayTracer: symbol lookup error: /home/anaitha2/vulkan-sim-root/mesa-vulkan-sim/lib/lib/x86_64-linux-gnu/libvulkan_lvp.so: undefined symbol: gpgpusim_allocBuffer

I was not facing this issue in my previous setups. What might be the possible reasons for this error and how do I resolve it? Any help is appreciated.

lucylufei, 2024-01-31:
Hi! 
The `gpgpusim_allocBuffer` function links buffer memory allocation between Mesa and Vulkan-Sim to ensure deterministic addressing. The function is defined [here](https://github.com/ubc-aamodt-group/vulkan-sim/blob/dc742faa6e5fcfaf415fe51eab71dcf9345388fd/src/cuda-sim/gpgpusim_calls_from_mesa.cc#L178) and is called from the Mesa driver. Please check that you have compiled the latest version of Vulkan-Sim with this function definition. Also check that you have sourced the `setup_environment` file so that this function can be found when executing the `RayTracer` application.

rodhuega, 2024-02-01:
Hi, I'm facing the same problem. I have done the setup_environment and I have that function in that file. Curiously, I have been able to run the simulator on a different machine. Both with Ubuntu 20.04.

lucylufei, 2024-02-05:
Hi, here are some other things to check:

Make sure that vulkan-sim compiled without errors.  After a successfully compilation, there should be a `libcudart.so` file inside the `./lib/gcc-[version]/cuda-[version]/release` folder associated with Vulkan-Sim. Please check that this file exists and that the gcc and cuda versions match your system. The symbol lookup error could arise if the gcc or cuda versions are updated after compilation. 

Another thing to check is the compiled Mesa driver. In the Mesa `lib` folder where `libvulkan_lvp.so` is found (the path from your error message), run `ldd libvulkan_lvp.so` to check if `libcudart.so` is successfully linked to the `libcudart.so` file from your compiled Vulkan-Sim. 

Lastly, you can update the `meson.build` file associated with the Mesa driver to explicitly point to the `libcudart.so` file. This can be set [here](https://github.com/ubc-aamodt-group/mesa-vulkan-sim/blob/b8f3e3980e87736fea176da6ebc81f82c6ea80e9/src/gallium/frontends/lavapipe/meson.build#L9) and then re-compile the Mesa driver. We find that setting this path here is unnecessary on our test systems, but it may help to resolve your error. 

Hope this helps!

rodhuega, 2024-02-07:
Hi, I have tried your suggestions, but I have not succeeded.

However, I have identified the problem, but I don't know how to solve it. The problem is that the first `ninja -C build/ install` is succeeding. Later on, I continued with all the steps, and in the end, it failed at the start of the execution with that error. I have checked `ldd libvulkan_lvp.so` and I have noticed that there isn't any `libcudart.so` link entry. So that I supposed that is the problem, but I don't know why is not linking. I have followed all the steps and suggestions.

PD: I have erased all the compilation files and tried to run meson and ninja with the reconfigure flag. Also, I have tried deleting the `lib` and `build` folders of the three repositories.

lucylufei, 2024-02-08:
The dynamic linking for `libcudart` should be in the same `meson.build` file [here](https://github.com/ubc-aamodt-group/mesa-vulkan-sim/blob/b8f3e3980e87736fea176da6ebc81f82c6ea80e9/src/gallium/frontends/lavapipe/meson.build#L10). Unfortunately, I'm not sure why the library is not linking for you. Does the embree library (`libembree3`) dynamically link to `libvulkan_lvp.so` properly? Since these libraries are both linked to `libvulkan_lvp.so` from the same `meson.build` file, we can try to narrow down the problem. 

Another thing you can try is adding the `link_args: ['-lcudart']` directly to `liblavapipe_st` [here](https://github.com/ubc-aamodt-group/mesa-vulkan-sim/blob/b8f3e3980e87736fea176da6ebc81f82c6ea80e9/src/gallium/frontends/lavapipe/meson.build#L68).

rodhuega, 2024-03-04:
I have found the issue. The problem was that CUDA was also installed through ``apt`` apart from being done from the ``.sh`` script.  Removing the ``apt`` installation fixed the issue. The reason when CUDA is also installed through ``apt``, `pkg-config` is finding the library in `/usr/lib/` and it is not doing properly the first `ninja -C build/ install`, so it is not trying to find the library with any of the paths that we are setting in Meson or as an environment variable.

Even though we have achieved fixing the bug in one machine due to removing the `apt` library, there is another machine with the same problem that we cannot remove because we are sharing that machine. It would be great if anyone came up with a solution. I'm not very familiar with Meson, so there is something that I can be missing. I suppose that there should be a way to force or prevent that a library is searched through `pkg-config`. I have searched information about that, but I have not found a way.

closed 2024-03-04 by anaitha2