ubc-aamodt-group/vulkan-sim #22: Running provided traces on cluster CPUs without a graphics card

haiyuem, 2024-03-08:
Hi, 

I was trying to run RayTracingInVulkan and vulkan-sim on a cluster with a CPU but without graphics cards. I don't need to generate new traces, just run the existing traces but change the GPGPUsim configurations in vulkan-sim. 

My command is: "./RayTracer --scene 18 --width 128 --height 128 --samples 2" as suggested in the README. 

I am getting the following error: 

> Vulkan Devices: 
> - [0] UnknownVendor 'llvmpipe (LLVM 12.0.0, 256 bits)' (CPU: vulkan 1.1.182, driver llvmpipe Mesa 21.2.6 (LLVM 12.0.0) - 0.0.1)
> 
> Setting Device [0]:
> FATAL: missing required extensions: VK_KHR_acceleration_structure, VK_KHR_deferred_host_operations, VK_KHR_ray_tracing_pipeline, VK_KHR_shader_clock
> 
>  0# 0x00005593AA3EDFDD in ./RayTracer
>  1# 0x00005593AA3EF1DD in ./RayTracer
>  2# 0x00005593AA432B3F in ./RayTracer
>  3# 0x00005593AA42D020 in ./RayTracer
>  4# 0x00005593AA43C8CD in ./RayTracer
>  5# 0x00005593AA44C9E6 in ./RayTracer
>  6# 0x00005593AA42DF56 in ./RayTracer
>  7# 0x00005593AA3F81B9 in ./RayTracer
>  8# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
>  9# 0x00005593AA3F93FE in ./RayTracer 

I think the missing extensions are because I don't have any graphics card available. However I read in one of the previous issues that the simulator should simulate the GPU ray tracing hardware and should run on any CPU. Do you have any clues on what I can do to fix this? Thank you!!

lucylufei, 2024-03-12:
Hi! If you are using the latest version of Vulkan-Sim, there should be no dependencies on any graphics card since the Lavapipe driver is a software implementation. From your error message, perhaps the program is not linking with the correct Vulkan ICD. Did you perform the `export VK_ICD_FILENAMES` step? This step ensures that Vulkan application will use the `libvulkan_lvp.so` generated by Vulkan-Sim instead of the default one on your PC. Thanks!

haiyuem, 2024-03-14:
Thank you!! I realized that I did mess up with that env variable. Now I can run the benchmarks :)

closed 2024-03-14 by haiyuem