ubc-aamodt-group/vulkan-sim #23: Running multiple jobs in parallel?

haiyuem, 2024-03-17:
Hi! This might be a dumb question - I was wondering if you have ever run multiple traces in parallel? If so, what is the set up? 

For example, I want to launch vulkan-sim runs for the benchmarks here together: 
https://github.com/ubc-aamodt-group/RayTracingInVulkan/blob/master/lumibench.sh

Each time I got different errors and the errors are not very self-explanatory. Most of them seem to point to concurrent writes to the same files. These errors never show up when I run them sequentially. 

I digged into the code a bit and found that both the gpgpusim code and the mesa-vulkan-sim have some hardcoded paths. For gpgpusim, I believe I can prevent concurrent writes by creating multiple directories that contain the RayTracer binary and save the stats to different places. However, for mesa-vulkan-sim conflicts, the code seems to be pointing to mesa-vulkan-sim/gpgpusimShaders, and errors come from unexpected results when other processors rewrite the files. Creating multiple gpgpusimShaders directories didn't help and brought me other errors. 

I was wondering if you know the best solution for this. Each run takes 20-30 mins on a single core so I'd love to speed them up by running multiple at once. Any help is appreciated!

tommychouyc, 2024-03-17:
What I've done in the past is to have a few seconds of delay between each job launch. After the shaders in gpgpusimShaders/ are created and parsed, they can be overwritten without any issue.

haiyuem, 2024-03-17:
Thank you for the really quick reply!! I just tried and it seems to work.

closed 2024-03-17 by haiyuem