ubc-aamodt-group/vulkan-sim #26: Do RT cores use general purpose ldst unit?

Yfeng-44, 2024-05-08:
Hi,

Thanks for you efforts! 

I'm wondering do rt cores use general purpose ldst unit?

In the [code](https://github.com/ubc-aamodt-group/vulkan-sim/blob/dc742faa6e5fcfaf415fe51eab71dcf9345388fd/src/gpgpu-sim/shader.cc#L3233), it seems that the rt unit is able to issue memory access on its own (unlike tensor core). Am I right on this?

If the rt core indeed does not use general purpose ldst unit, could you please provide some reference (if there is any)?

Thanks

tommychouyc, 2024-05-09:
We modelled the RT unit like a LDST unit that can issue its own memory accesses.
In the [Nvidia Ampere whitepaper](https://www.nvidia.com/content/PDF/nvidia-ampere-ga-102-gpu-architecture-whitepaper-v2.pdf) table 4 (page 18,19), it says the RT cores have a dedicated L1 interface (third row from the bottom) which probably implies that it can access memory independently. I think this is the closest we'll get to public information.

Hopefully that helps!

Yfeng-44, 2024-05-09:
Thank you! This helps!

closed 2024-05-09 by Yfeng-44