ubc-aamodt-group/vulkan-sim #27: Support for CUDA apps

Yfeng-44, 2024-05-18:
Hi vulkansim team!

I noticed that in Lumibench paper, there is a comparison section that run Rodinia application. However, it seems that the functionality of simulating general purpose application is bit broken with my fresh build of vulkan-sim.

I guess it is because vulkan-sim bind and read/write memory in a different way than original gpgpu-sim at https://github.com/ubc-aamodt-group/vulkan-sim/blob/dc742faa6e5fcfaf415fe51eab71dcf9345388fd/src/cuda-sim/memory.cc#L165. 

Can you please verify it?

Thanks!

tommychouyc, 2024-05-18:
Hi, that use_external_launcher flag has not been used since we published LumiBench. It was only used back when the simulator was first published and only supported the Intel driver. Could you run this in debug mode and show us where the simulator crashes when running GPGPU workloads?

Yfeng-44, 2024-05-18:
Hi, 

I source'ed environment setting with debug and run with LUD application in Rodinia (also used in lumibench paper). Here are the output:

`GPGPU-Sim PTX: instruction assembly for function '_Z12lud_internalPfii'...   done.
GPGPU-Sim PTX: finished parsing EMBEDDED .ptx file lud-rodinia-3.4.sm_62.ptx
GPGPU-Sim PTX: loading globals with explicit initializers... 
GPGPU-Sim PTX: finished loading globals (0 bytes total).
GPGPU-Sim PTX: loading constants with explicit initializers...  done.
GPGPU-Sim PTX: Loading PTXInfo from lud-rodinia-3.1.sm_60.ptx
GPGPU-Sim PTX: Loading PTXInfo from lud-rodinia-3.3.sm_60.ptx
GPGPU-Sim PTX: Kernel '_Z12lud_internalPfii' : regs=23, lmem=0, smem=2048, cmem=336
GPGPU-Sim PTX: Kernel '_Z13lud_perimeterPfii' : regs=32, lmem=0, smem=3072, cmem=336
GPGPU-Sim PTX: Kernel '_Z12lud_diagonalPfii' : regs=32, lmem=0, smem=1024, cmem=336
GPGPU-Sim PTX: Loading PTXInfo from lud-rodinia-3.2.sm_62.ptx
GPGPU-Sim PTX: Loading PTXInfo from lud-rodinia-3.4.sm_62.ptx
GPGPU-Sim PTX: Kernel '_Z12lud_internalPfii' : regs=23, lmem=0, smem=2048, cmem=336
GPGPU-Sim PTX: Kernel '_Z13lud_perimeterPfii' : regs=32, lmem=0, smem=3072, cmem=336
GPGPU-Sim PTX: Kernel '_Z12lud_diagonalPfii' : regs=32, lmem=0, smem=1024, cmem=336
GPGPU-Sim PTX: __cudaRegisterFunction _Z13lud_perimeterPfii : hostFun 0x0x563b94ccf580, fat_cubin_handle = 2
GPGPU-Sim PTX: __cudaRegisterFunction _Z12lud_diagonalPfii : hostFun 0x0x563b94ccf740, fat_cubin_handle = 2
WG size of kernel = 16 X 16
Generate input matrix internally, size =256
Creating matrix internally size=256
Could not find 0xc0000000 in Vulkan address map
gpgpusim: WARNING: Memory backing buffer not found for address 0xc0000000. This data write may be invalid
Segmentation fault (core dumped)`

tommychouyc, 2024-05-18:
I think as long as the program doesn't use any of the ray tracing functions we added, you can set the use_external_launcher flag to true and it should restore the original gpgpusim behavior.

https://github.com/ubc-aamodt-group/vulkan-sim/blob/dc742faa6e5fcfaf415fe51eab71dcf9345388fd/src/cuda-sim/vulkan_ray_tracing.cc#L120

Yfeng-44, 2024-05-18:
Hi thanks for replying fast,

I also tried to change this flag. but it still doesn't work. log is attched.

```GPGPU-Sim PTX: instruction assembly for function '_Z12lud_internalPfii'...   done.
GPGPU-Sim PTX: finished parsing EMBEDDED .ptx file lud-rodinia-3.4.sm_62.ptx
GPGPU-Sim PTX: loading globals with explicit initializers... 
GPGPU-Sim PTX: finished loading globals (0 bytes total).
GPGPU-Sim PTX: loading constants with explicit initializers...  done.
GPGPU-Sim PTX: Loading PTXInfo from lud-rodinia-3.1.sm_60.ptx
GPGPU-Sim PTX: Loading PTXInfo from lud-rodinia-3.3.sm_60.ptx
GPGPU-Sim PTX: Kernel '_Z12lud_internalPfii' : regs=23, lmem=0, smem=2048, cmem=336
GPGPU-Sim PTX: Kernel '_Z13lud_perimeterPfii' : regs=32, lmem=0, smem=3072, cmem=336
GPGPU-Sim PTX: Kernel '_Z12lud_diagonalPfii' : regs=32, lmem=0, smem=1024, cmem=336
GPGPU-Sim PTX: Loading PTXInfo from lud-rodinia-3.2.sm_62.ptx
GPGPU-Sim PTX: Loading PTXInfo from lud-rodinia-3.4.sm_62.ptx
GPGPU-Sim PTX: Kernel '_Z12lud_internalPfii' : regs=23, lmem=0, smem=2048, cmem=336
GPGPU-Sim PTX: Kernel '_Z13lud_perimeterPfii' : regs=32, lmem=0, smem=3072, cmem=336
GPGPU-Sim PTX: Kernel '_Z12lud_diagonalPfii' : regs=32, lmem=0, smem=1024, cmem=336
GPGPU-Sim PTX: __cudaRegisterFunction _Z13lud_perimeterPfii : hostFun 0x0x55a1783b7580, fat_cubin_handle = 2
GPGPU-Sim PTX: __cudaRegisterFunction _Z12lud_diagonalPfii : hostFun 0x0x55a1783b7740, fat_cubin_handle = 2
WG size of kernel = 16 X 16
Generate input matrix internally, size =256
Creating matrix internally size=256
Segmentation fault (core dumped)```

tommychouyc, 2024-05-18:
Could you also attach the gdb backtrace? Thanks!

Yfeng-44, 2024-05-18:
Attached, Thank you so much!!

```
(gdb) bt
#0  0x00007f7d1c49d9fb in std::__cxx11::list<stream_operation, std::allocator<stream_operation> >::begin (this=0x8)
    at /usr/include/c++/9/bits/stl_list.h:942
#1  0x00007f7d1c49d875 in std::__cxx11::list<stream_operation, std::allocator<stream_operation> >::front (this=0x8)
    at /usr/include/c++/9/bits/stl_list.h:1111
#2  0x00007f7d1c49d6f6 in CUstream_st::front (this=0x0) at stream_manager.h:235
#3  0x00007f7d1c49c911 in stream_manager::register_finished_kernel (this=0x55e6c3f28210, grid_uid=3287445632) at stream_manager.cc:293
#4  0x00007f7d1c49c8c6 in stream_manager::register_finished_kernel (this=0x55e6c3f28210) at stream_manager.cc:286
#5  0x00007f7d1c48dddb in gpgpu_sim_thread_concurrent (ctx_ptr=0x55e6c3f0e190) at gpgpusim_entrypoint.cc:155
#6  0x00007f7d1b6e4609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#7  0x00007f7d1ba39353 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

lucylufei, 2024-05-18:
Hi, thanks for sharing all the details! There were a few changes that we made to run Rodinia workloads which weren't included with the Vulkan-Sim release. I will sort out those changes and get back to you in a couple of days :)

Yfeng-44, 2024-05-18:
Hi @lucylufei ,

Thank you so much!
Have a good weekend!

lucylufei, 2024-05-22:
Hi @Yfeng-44, I've pushed the changes to a separate cuda branch. The `use_external_launcher` variable has been moved into the `gpgpusim.config` file so please add `-gpgpu_rt_external_launch 1` to the configs when running CUDA applications. You can change this back to `-gpgpu_rt_external_launch 0` for Vulkan applications. You may also need to modify the `-gpgpu_max_simulated_rt_kernels` parameter to simulate multiple kernel launches on CUDA applications. 

Please let me know if you still encounter any issues.

Yfeng-44, 2024-05-23:
Hi @lucylufei ,

Thanks for the efforts! I will try the change. 

Will close this issue for now, and will let you know if there's any issues.

Thanks!

closed 2024-05-23 by Yfeng-44