ubc-aamodt-group/vulkan-sim #28: Program Aborts or Gets killed with Modified Parameters

quanzw99, 2024-05-30:
Hi Vulkan-sim Team,

I am attempting to change parameters to generate more primitive rays, but I am encoutering `abort` and `killed` errors.

Here aer the two scenarios I encountered:

1. When I run `./RayTracer --scene 18 --width 128 --height 128 --samples 2 > PARTY_PT.log`, the PPM file is generated successfully, completing in about 54 minutes on my local environment. However, when I change the parameters to `./RayTracer --scene 18 --width 256 --height 256--samples 2 > BUNNY_PT_w256.log`, the program gets killed after a few hours, and the log file is incomplete. I have attatched the log [PARTY_PT_w256.log](https://github.com/ubc-aamodt-group/vulkan-sim/files/15503786/PARTY_PT_w256.log). It seems to be stuck somewhere.
[image]

2. When I run `./RayTracer --scene 30 --width 128 --height 128 --samples 8 > REF_PT.log`, the program aborts due to an assert failure. I have attached a sreenshot of it. It seems to have a complete log file, but I am concerned that the assert failure may have prevented all the rays from being fully generated.

[image]

I would greatly appreciate any insights or clarifications.
Thank you very much for your time and assistance.

quanzw99, 2024-05-30:
https://github.com/ubc-aamodt-group/vulkan-sim/issues/18 It seems this issue has already solved my first problem.

lucylufei, 2024-05-30:
Hi @quanzw99, can you please share the full log for REF_PT? This assertion error is related to copying the frame buffer so it shouldn't affect the ray generation, however, I'd like to see the full log to confirm. Thanks!

quanzw99, 2024-05-31:
Yes. [REF_PT.log](https://github.com/user-attachments/files/15509813/REF_PT.log)
I also attached the BUNNY_PT, which meet the same assertion error using `./RayTracer --scene 18 --width 128 --height 128--samples 8 > BUNNY_PT.log`. [BUNNY_PT.log](https://github.com/user-attachments/files/15509841/BUNNY_PT.log)

Thank you for your response!

lucylufei, 2024-05-31:
Thanks for sharing the logs! I don't see the assertion error in either log, but there should be no problems with your simulation based on what I do see. If you get the full GPGPU-Sim output with all the statistics and see the following, then the kernel for the current frame is complete. 

```
Max simulated kernels of 1 has reached, exiting.
GPGPU-Sim: *** exit detected ***
```

If you're still worried about rays, you can check the `rt_n_closesthit_rays` stat, which reports the total number of rays that were simulated. This total is based on the image size, the number of samples per pixels, and the number of bounces per sample.

quanzw99, 2024-05-31:
Yes, although I have redirected the logs to a log file, the assertion errors are displayed in the console output, which is why you didn't see them in the log. However, from the log file, they are indeed complete.

Thank you for informing me that by checking the `rt_n_closesthit_rays` data, we can know the total number of rays. I still have a question: what potential impact could this assertion errors have?

Thanks again!

closed 2024-06-01 by quanzw99