ubc-aamodt-group/vulkan-sim #31: Whether to support coherent engine?

quanzw99, 2024-08-23:
Hi Vulkan Sim Team,

I modified the configuration to enable the coherent engine, but encountered an error: `ray_coherency_engine.cc:179: void ray_coherence_engine::cycle(): Assertion 'm_num_ray_pool_rays + m_num_scheduled_rays == m_total_rays' failed.`, when I run the command `./RayTracer --scene 18 --width 128 --height 128 --samples 2`. I would like to confirm whether the current version of Vulkan-sim supports the coherent engine (in other words, is this error expected?).

Thank you for your time and assistance!

tommychouyc, 2024-08-23:
Hi, the coherency engine is not supported. It was a small experiment that we tried early on during development and hasn't been updated since then.

quanzw99, 2024-08-23:
Thank you for your prompt response!

closed 2024-08-23 by quanzw99