ubc-aamodt-group/vulkan-sim #35: How to add a new scene?

rodhuega, 2024-10-10:
I would like to know what is needed and how to add a new scene to Vulkan-sim. Is there any guide about that?

lucylufei, 2024-10-10:
You can add new scenes [here](https://github.com/ubc-aamodt-group/RayTracingInVulkan/blob/f5932e7e3a114f22ba0cd6ef58d17887e56f9af6/src/SceneList.cpp#L68) and adding a corresponding function to load the scene. You can use the [BUNNY scene](https://github.com/ubc-aamodt-group/RayTracingInVulkan/blob/f5932e7e3a114f22ba0cd6ef58d17887e56f9af6/src/SceneList.cpp#L337) as a reference, which loads an `.obj` file, transforms it, and sets up the other necessary scene parameters.

rodhuega, 2024-10-11:
Thanks for the help.

closed 2024-10-11 by rodhuega